COVESA/Open1722 #20: Aggregation of multiple CAN messages in 1722 frame

adriaan-niess, 2024-07-25:
Based on comment from @JLPLabs we should consider adding support to aggregate multiple CAN frames into a single Ethernet frame.

> Adriaan,
> You'll see a 3rd commit on this PR. 
> 
> NOTE  PLEASE LET ME KNOW:
> a) if this is more than you want,
> b) if there is a better / more idiomatic approach I should be taking.
> 
> This last change can be summarized as:
> * LISTENER
>   * updated one line in `acf-can-listener` so that the `acf_pdu` pointer is moved through the ethernet frame. (right now there is logic that seems to imply that was the desired behavior, but the pointer wasn't actually being updated).
> 
> * TALKER
>   * added a `--count COUNT` flag to the command line. This allows the user to require COUNT number of CAN frames to be packed into one Ethernet frame.
>   * used the variable that already existed, `num_acf_msgs` for handling the count logic.
> 
> _Originally posted by @JLPLabs in https://github.com/COVESA/Open1722/issues/18#issuecomment-2248463892_

adriaan-niess, 2024-07-26:
Hey, this looks very well thought out to me. Big thanks! Unfortunately I can only give you a short reply, because I'm in a little bit of hurry. Some thoughts:

- Seems plausible to me now that you might want to limit the number of CAN frames for debugging purposes :+1: 
- I think it makes more sense to interpret the timeout with respect to the oldest (not the newest) message in the buffer. The reason is that you want to be able to guarantee a deadline (or worst-case transmission time) for your CAN messages. For example you might want to ensure that certain sensor signals are delivered to your central compute in under <100ms. If you reset the timeout every time a new CAN message is added to the buffer, it becomes quite hard to estimate for how long CAN frames could get delayed in the worst-case.
- Having ~1h as the max timeout seems more than enough to me. I don't think there are use cases where it's acceptable to wait 1h to receive the data even it's not that important. I'd estimate that in most real-world scenarios the timeout would be set to <1s.

JLPLabs, 2024-07-29:
Sorry for my delayed response. 
I think your logic for timing makes sense. I'll start to work on updated code with this definition for ``--timeout``:

* ``--timeout <time>`` where ``<time>`` is an integer number of microseconds, "usec". (*not* milliseconds)
  *  The timeout establishes the maximum wait for retransmission of a CAN message.
  * The timeout clock starts with the arrival of the first message.
  *  The retransmission may be sooner than ``<time>`` usec if the Ethernet frame becomes full or the maximum count (see ``--count``) of CAN messages has arrived.
  * minimum value: 0
  * maximum value: (2^32 - 1)
  * default: maximum value
  
   examples:
    * ``--timeout 0`` means no waiting, immediately pack and send the CAN frame in a 1722 frame.
    * ``--timeout 700`` means wait no longer than 700 usec from the arrival of the first CAN frame to start the 1722 frame transmission process. 
    * if no value for ``--timeout`` is given, then the 1722 frame will be launched when one of these is true:
      * the 1722 frame is full
      * the 1722 frame has ``<count>`` number of messages
      * 1.193 hours has passed since the arrival of the first message (in a real network this effectively never going to happen)

adriaan-niess, 2024-07-30:
Hi,

don't worry take your time.

- Only thing I'd change is to set the default timeout to 0 instead of 2^32-1 because then it could happen that a single CAN frame gets delayed for a very long time. I fear some users could interpret this as a bug if they don't study the documentation first.

If you need any help or feel stuck at some point let me know.

Regards
Adriaan