COVESA/Open1722 #90: Is expanding the capabilities of the acf-can examples desirable?

JLPLabs, 2025-04-09:
This is more of a discussion item, so sorry it appears as an "issue". (if there is a better mechanism for this kind of topic please let me know!)

The AEF Highspeed ISOBUS project [^1] is likely to do CAN tunneling with some capabilities not found in the current version of the examples for acf-can. These new capabilities **might** be of interest to the wider Open1722 community.

- optional use of VLAN
- optional use of IPv6 UDP
- optional use of CAN Brief (ACF message type 0x02)

If they are of interest then I'm willing to share them here. 

[^1]: https://www.aef-online.org/about-us/teams.html#/ExpertTeams

nayakned, 2025-04-14:
@JLPLabs We are in general happy to accept any contributions that may be of use to the wider community. However, we would like to keep the examples rather simple so as to serve as a template for applications using Open1722. We can think of making multiple examples where, e.g., VLAN tags or an IPv6 encapsulation is the focus. Since we have now refactored many common parts of the ACF CAN conversion to __acf-can-common.c__, it should be quite possible to do something like acf-can-vlan-listener/talker/bridge etc.

Any other opinions? @SebastianSchildt @adriaan-niess

nayakned, 2025-04-14:
For contributing to this repository, please also adhere to the [contribution guidelines](https://github.com/COVESA/Open1722/blob/main/CONTRIBUTING.md).

adriaan-niess, 2025-04-14:
Hi @JLPLabs ,

expanding the examples is definitely desirable. Unfortunately we haven't been able to merge some of your previous work because we diverged quite a lot in the mean time. I'm very sorry about that! Fortunately I think especially the CAN example(s) should now be relatively stable. Overall I'd suggest to keep changes small (if possible) and try to split bigger proposals into multiple smaller ones.

I do like your suggestions. But shouldn't VLANs already work out-of-the-box on Linux? They way they work in Linux is to create a new Network Interface that could then be used by one of the example applications to send/receive 1722 frames. Or do you have a use that is not covered by that?

Regarding the ISOBUS project, is the use of the IEEE 1722 protocol considered here? I'm going to have a look in case I can access the member area. The current revision of IEEE1722 should btw be released in Q2 and is extending the existing CAN formats, so there will be a CAN-V2 format and an extra format for CAN-XL soon and we hope to add the new formats.

Regards
Adriaan

JLPLabs, 2025-04-15:
All,
It sounds like you are open to the concepts listed in the initial comment. That is most welcome. Thank you!

It is good to know that Linux can provide the VLAN tagging. I'll defer to my colleagues if they want a standard approach in C for the cases where an ECU isn't running Linux. 

So, to start with, I'll create two separate examples:
1)  **udp6** -  `Open1722/examples/acf-can/linux/acf-can-udp6-{listener|talker|bridge} `
2)  **Brief** - `Open1722/examples/acf-can/linux/acf-can-brief-{listener|talker|bridge} `

Any suggestions or concerns?

-John

JLPLabs, 2025-04-15:
So, I have a philosophical question for you... (assuming that you want `acf-can-common.{h|c}` updated to support "CAN Brief").

Is it better to: 
* increase complexity by adding a function `can_to_avtp_brief`
* break current examples by adding a parameter to `can_to_avtp` that indicates use of CAN Brief format?

nayakned, 2025-04-17:
My initial thoughts: 

1. I would add extra functions to ```acf-can-common.{h|c}``` for acf-can-brief. acf-can-brief shares very little code/API with acf-can due to the header structure of IEEE 1722. So trying to wrap it in a single function is not giving any additional benefit in my opinion.

3. I am not sure if we need an application specifically for UDP6. On Linux, depending on the use of the interface/IP address used this should automatically happen just like how VLANs are handled by the interface assignment. That being said I think it is a good idea to refactor the current listener/talker/bridge to remove UDP and make a UDP specific application. So overall we would have maybe 2-3 examples. 
- acf-can-udp-examples
- acf-can-ethernet-examples
- acf-can-brief-examples
Then depending on the network configuration, we can use combinations of IPv4/IPv6/VLANs etc.

JLPLabs, 2025-04-18:
I understand.

Before I get too far, do these two function declarations match your expectations? (**NOTE: UDP has been removed**).

Functions added `acf-can-common.h`:

1.  `avtp_brief_to_can`
2. `can_to_avtp_brief`

```
/**
 * Function that converts AVTP CAN Brief Frames to CAN
 *
 * @param pdu: Start of the AVTP Frame
 * @param can_frames: Array of CAM Frames to be recovered from AVTP CAN Brief Frames
 * @param can_variant: AVTP_CAN_CLASSIC or AVTP_CAN_FD
 * @param stream_id: AVTP stream ID of interest
 * @param exp_cf_seqnum: Expected Control format sequence num.
 * @return Number of CAN messages received
 */
int avtp_brief_to_can(uint8_t* pdu, frame_t* can_frames,
                      Avtp_CanVariant_t can_variant, uint64_t stream_id,
                      uint8_t* exp_cf_seqnum);

/**
 * Function that converts CAN Frames to AVTP CAN Brief Frame
 *
 * @param can_frames: Array of CAM Frames to be translated to AVTP Frames
 * @param can_variant: AVTP_CAN_CLASSIC or AVTP_CAN_FD
 * @param pdu: Start of AVTP Frame
 * @param use_tscf 1: TSCF, 0: NTSCF
 * @param stream_id: AVTP stream ID of interest
 * @param num_acf_msgs: No. of ACF CAN messages to aggregate
 * @param cf_seq_num: Control format sequence num.
 * @return Length of the PDU
 */
int can_to_avtp_brief(frame_t* can_frames, Avtp_CanVariant_t can_variant, uint8_t* pdu,
                      int use_tscf, uint64_t stream_id,
                      uint8_t num_acf_msgs, uint8_t cf_seq_num);
```

nayakned, 2025-04-22:
This looks good. We will need then additional functions to add and remove UDP bindings, but that is ok.
Just as a note: These functions are also used in the Zephyr applications [acf-can-bridge](https://github.com/COVESA/Open1722/blob/main/examples/acf-can/zephyr/acf-can-bridge.c#L204). These applications would also need to be modified.

SebastianSchildt, 2026-09-13:
I will close this for now. There is a CAN_BRIEF talker/listener/bridge now. VLAN I think is handled Linux side. For UDP6 (nobdoy ever teted I guess) or other improvements feel free to open another issue/PR

closed 2026-09-13 by SebastianSchildt