COVESA/Open1722 #97: API Enhancement: Rework set/get length of ACF types

adriaan-niess, 2025-05-29:
Using APIs like `Avtp_Can_SetAcfMsgLength(...)` is error prone since the length is encoded in quadlets. @nayakned We discussed that quite a while ago.

I'd propose to keep only `Avtp_AcfCommon_SetAcfMsgLength(...)` and remove `Avtp_<Type>_SetAcfMsgLength(...)` for all ACF types and instead provide a function `Avtp_<Type>_SetPayloadLen(...)` which uses bytes instead of quadlets and also implicietly sets the correct padding.

Analog to removing `Avtp_<Type>_SetAcfMsgLength(...)`, it would then also possible to remove `Avtp_<Type>_SetPad(...)` because the padding would then implicitely be set by calling `Avtp_<Type>_SetPayloadLen(...)`.

To summarize, I propose to:

- Remove `Avtp_<Type>_SetAcfMsgLength(...)` for all ACF types (except AcfCommon)
- Remove `Avtp_<Type>_GetAcfMsgLength(...)` for all ACF types (except AcfCommon)
- Remove `Avtp_<Type>_SetPad(...)` for all ACF types (except AcfCommon)
- Remove `Avtp_<Type>_GetPad(...)` for all ACF types (except AcfCommon)
- Add `Avtp_<Type>_SetPayloadLen(...)` to all ACF types (except AcfCommon)
- Add `Avtp_<Type>_GetPayloadLen(...)` to all ACF types (except AcfCommon)

This might cause some breaking changes, so the earlier we follow through with this, the better. In case we want to follow through with this, I'd say we should also add a new version tag to the main branch.

@nayakned and @SebastianSchildt what are your thoughts?

Jignesh-007, 2025-05-30:
Hello @adriaan-niess, @nayakned  and @SebastianSchildt 
I would recommend to deprecate the  apis Avtp_<Type>_SetAcfMsgLength(...), Avtp_<Type>_GetAcfMsgLength(...), Avtp_<Type>_SetPad(...), Avtp_<Type>_GetPad(...) and add new one.
In readme file mark this as deprecate and mention when it will be removed.
After this time frame we can safely remove the api's
this time frame can be 6 Months or 1 Year.

adriaan-niess, 2025-05-30:
@Jignesh-007 deprecating sounds reasonable. This way there's no need to touch existing code. Overall less work which I like.

SebastianSchildt, 2025-05-30:
Just ran into length pain yesterday in a project :) I assume we would still need `Avtp_<Type>_GetAcfMsgLength`(in bytes PLEASE?) to be able to set the length in (N)TSCF header correctly?

adriaan-niess, 2026-08-21:
Summary after taking feedback into account

- Remove `Avtp_<Type>_SetAcfMsgLength(...)` for all ACF types (except AcfCommon)
- Remove `Avtp_<Type>_GetAcfMsgLength(...)` for all ACF types (except AcfCommon)
- Remove `Avtp_<Type>_SetPad(...)` for all ACF types (except AcfCommon)
- Remove `Avtp_<Type>_GetPad(...)` for all ACF types (except AcfCommon)
- Add `Avtp_<Type>_SetPayloadLength(...)` to all ACF types (except AcfCommon), set payload length in bytes (implicitely acf_msg_length and pad field)
- Add `Avtp_<Type>_GetPayloadLength(...)` to all ACF types (except AcfCommon), returns payload length in bytes
- Add `Avtp_<Type>_GetMessageLength(...)` to all ACF types, returns message length including header in bytes

nayakned, 2026-08-24:
I hate to nitpick but my issues with the two proposals are following.

- While I agree we can break the API (better now than later), `GetAcfMsgLength()` and `SetAcfMsgLength()` is a good (at least a spec conform) API. The specification calls this field _acf_msg_length_. I agree the API does not directly mention that it works with quadlets, but we can mention that in the API documentation. Any other helper API can explicitly mention in the function name that it works with octets/bytes. So I would vote against a `SetAcfMessageQuadLength()`. We can perhaps think of adding a hint in the API names to say its a convenience function (and can hence differ in naming from the spec) versus the names/semantics of the fields are taken as is from the spec.
- 100% agreed finalize is a dumb word. But when we use `Avtp_<type>_GetPayloadLength` and `Avtp_<type>_SetPayloadLength` are we referring to the payload of the message or the payload of the _type_, e.g., CAN payload? We need APIs for both perhaps.

When the initial versions of Open1722 were developed, not a lot of thought has gone into this.

SebastianSchildt, 2026-08-24:
Discussion Aug 24

- **Keep** `Avtp_<Type>_SetAcfMsgLength(...)`  - Sets message length in quadlets
- **Keep** `Avtp_<Type>_GetAcfMsgLength(...)`  - Gets message length in quadlets
- **Keep** `Avtp_<Type>_SetPad(...)` - Set padding bits
- **Keep** `Avtp_<Type>_GetPad(...)` - Get padding bits
- **Add** `Avtp_<Type>_SetPayloadLength(...)` - Set payload length in bytes (will calculate and set AcfMsgLength is quads and set pad field as required and set pad bytes to 0)
- **Add** `Avtp_<Type>_GetPayloadLength(...)` - Get payload length in bytes, without padding 
- **Add** `Avtp_<Type>_GetAcfMsgLengthInBytes(...)` - Get message length (header + payload+pad) in bytes

adriaan-niess, 2026-08-24:
Hi @nayakned , the [idea](https://github.com/COVESA/Open1722/issues/97#issuecomment-5394259423) was to name it `GetAcfMsgQuadLength()` instead of `GetAcfMsgLength()` because people will probably not frequently use this function and are way more likely to use the `Avtp_<Type>_GetAcfMsgLengthInBytes(...)` function. Therefore the idea was to give the function that is used more often a more convenient name and the function which is used less often a less convenient name.

Nonetheless I'd also be happy to move forward with the latest [suggestion](https://github.com/COVESA/Open1722/issues/97#issuecomment-5396132478) from @SebastianSchildt from August 24

SebastianSchildt, 2026-09-02:
Closing this as done by the tickets in #166

Still a follow up in #182

closed 2026-09-02 by SebastianSchildt