COVESA/Open1722 #105: Rework field code to avoid functions calls

go2sh, 2025-08-25:
Hi @all,

I really like your library and it helps us developing new apps very fast.

One problem I faced with it, especially on IFX tricore devices is, that every field get/set API is a function call. This adds a huge overhead: 
 * The instruction cache is poisened
 * Add for at least 4 additional cycles for the call it self
 * Increases the code size  because most field getters/setter can be render in a single instruction if the data is loaded.

This could be fixed relatively easy by making all getters setters inline.

BR,
go2sh

SebastianSchildt, 2025-08-25:
It makes sense, as the actual code that needs to be executed is small, so function call overhead is visible. I am just wondering, as the set/gets are really simple abstractions over the field setting Macros, I would have assumed the compiler might inline it in terms of optimization? 
Did you check whether that has any effect, or is the issue that you build with -Os or some equivalent for code size and still want to force inlining?

I am wondering we should "just inline" and consider it an improvement for any user (like we did when we "const" more), or whether we still should hide it behind a #define, because it might have unintended consequences?

go2sh, 2025-08-26:
I was also creating some stress tests comparing tc3 and tc4. And I realized that the IPC was very low with the library code. For can Brief, i did a Prototype and i got a 20-30% IPC increase.

nayakned, 2026-05-06:
@SebastianSchildt @adriaan-niess Can this issue be closed?

SebastianSchildt, 2026-09-02:
With the changes in #166  now all ACFs (except VSS) including NTSCF/TSCF functions are fully inline. Closing this now, for specific ideas what to change/inline more feel free to open another issue (or re-open this one )

closed 2026-09-02 by SebastianSchildt